OpenRoberta/openroberta-lab #223: HiTechnic NXT Color Sensor Version 2 support for EV3

bjost2s, 2019-08-15:
**Is your feature request related to a problem? Please describe.**
 - A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
 - A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
 - A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
 - Add any other context or screenshots or a mockup about the feature request here.

simonedegiacomi, 2019-08-19:
@BeateJ @boonto  Specifically, which functions do we want to provide for this sensor?
These are the functions available through EV3-G:
[image]

boonto, 2019-08-19:
We tried it today, this should be the mode mapping:
- Color as colour mode
- W from RGB as light mode
- W from Passive as ambient light mode
- RGB as RGB mode
I just added the blockly block, however found we already had something implemented for NXT.
This wasnt that nice so I changed stuff slightly, but we have to discuss this again.
That is why Ill push my small changes to your c4ev3 branch so we dont commit half finished stuff.
You should be able to work on the rest.

boonto, 2019-08-22:
Tested on c4ev3, ev3dev and lejos.

boonto, 2019-11-13:
I forgot adding the wait untils, this leads to a server error.
Todo:
- [ ] Add wait until for HT sensor modes

bjost2s, 2020-01-04:
There is still an issue with the codegeneration and checker, if there is the common colour sensor in the configuration connected instead of the HT Colour sensor:
here is an example with port 1 (common Colour Sensor) and port 2 (HT Colour Sensor) 
[image]

boonto, 2020-01-06:
At the time of implementing this, we had decided to reuse the `ColorSensor` for this, to avoid creating a redundant `HTColorSensor`. However, this leads to problems like mentioned above with worse code with multiple `if (block is ht color)` clauses to catch every different behaviour.

I have now added the specific `HTColorSensor` which, although duplicating most of `ColorSensor` initially, leads to a much cleaner implementation.

closed 2020-01-07 by bjost2s