OpenRoberta/openroberta-lab #224: Serial Write should show 'not supported'

rhasel4s, 2019-08-16:
**Describe the bug**
 When using the simulation the serial write block is not supported. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to any robot with serial write
2. Add the block to the program
3. Run the program in simulation

**Expected behavior**
 - The user should be informed, that the serial write block is not supported in simulation

**Screenshots**
 
[image]

ensonic, 2019-10-13:
I think we could support serial write. We could have a text view in the simulation that shows the text output from serial write.

bjost2s, 2024-09-24:
This feature is already implemented, see developer console:
[image]

closed 2024-09-24 by bjost2s