OpenRoberta/openroberta-lab #225: micro:bit create list shows wrong type

rhasel4s, 2019-08-16:
**Describe the bug**
 - when creating a list via the advanced block menu the drop-down shows the option colour, which is not supported by micro:bit.

**To Reproduce**
1. Go to micro:bit
2. open the advanced menu
3. drag on of the two or both create list blocks to the workspace
4. click on the drop-down menu

**Expected behavior**
 - The drop-down menu should show the same options like when creating the list via the variables option (see first screen)
**Screenshots**
 
[image]
[image]
[image]

practiceiqx, 2019-08-16:
？？？

boonto, 2019-08-28:
Types are now consistent.

closed 2019-08-28 by boonto