OpenRoberta/openroberta-lab #229: Remove ping request whenever it is not needed

bjost2s, 2019-08-22:
The ping request is only needed when the client would like to get information about the state of the robot. Depending on the way the programs are transferred to the system, the ping might not send back any information. The robot.connection property can have the following values: arduinoAgentOrToken, token, autoConnection, webview and local. AutoConnection, webview and local are applied to systems where the state of the robot is always unknown (for the server) so no ping is needed.
@VinArt, you have introduced "agent" as an additional property on the client site, but there is currently no plugin using it, so it will be removed with this commit.

bjost2s, 2019-08-23:
Due to the reduced httpsession timeout (60) we have to keep the ping. But systems which do not need it should ping less often.

bjost2s, 2019-11-06:
When switching the robot system, sometimes the new ping time takes too much time, blocked by the current timeout.

boonto, 2019-11-08:
After running programs on Dev/Lejos the run button does not reactivate.

boonto, 2019-11-08:
Run button works for EV3 again.
Run button is not reactivated randomly during compilation process, of e.g. Arduino, anymore.

closed 2019-11-08 by boonto