OpenRoberta/openroberta-lab #230: Replace remaining usages of BlocklyMethods

boonto, 2019-08-23:
Continuation issue to partially solved issue #95:
`BlocklyMethods` and `NEPODefs` still have to be removed from Java and C based robots.

- [x] Find all "BlocklyMethods.py" / "NEPOdefs.h" etc.
- [x] Find common methods
- [ ] Extract one liners (constants?) directly into code generation
  - [ ] Java
  - [ ] C
- [ ] Combine multi liners into common helper methods
  - [ ] Java
  - [ ] C
- [ ] Implement class in all robots
  - [ ] Java
  - [ ] C
- [ ] Default implementations for all languages
- [ ] Special implementations for robots that don't support default
- [x] Find way to parse methods from helper text file (YAML)
- [ ] Unit tests
  - [x] YAML override
  - [ ] Java robots
  - [ ] C robots
  - [ ] Helper itself?
- [ ] Test each method on robots
  - [ ] Java robots
  - [ ] C robots

boonto, 2019-10-24:
Will be continued in #278.

bjost2s, 2019-11-08:
couldn't find any string "BlocklyMethods" in the java code generation files.

closed 2019-11-08 by bjost2s