OpenRoberta/openroberta-lab #237: After computer suspend: This browser tab is not valid anymore...

Digitalwerkstatt, 2019-08-30:
**Describe the bug**
When I return to an openroberta-lab session after a couple of minutes of computer suspend / sleep, it is not possible to compile the program anymore. Instead a message tells me "This browser tab is not valid anymore...". Though, there are no other tabs opened. Reloading the page fixes the problem but the program is lost unless it is exported before.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'https://lab.open-roberta.org'.
2. Create a simple program, e.g. "Start, Show Text 'Hello'".
3. Put the computer to suspend / sleep.
4. Wake up the computer 5 minutes later.
5. Click on the play button to compile the program.

**Expected behavior**
 - The nepoprog compiled program is offered for download.

**Actual behavior**
 - A message stating "This browser tab is not valid anymore..." pops up.
- The play button is grayed out.

**Screenshots**
[image]

**Device information**
 - Type Desktop
 - OS: Mac OS X 10.14.6
 - Browser Safari 12.1.2 (14607.3.9) and Chrome Version 76.0.3809.100

**Additional context**
 - There are no other browser tabs opened in the background as the message suggests.

rbudde, 2019-09-06:
solved by extending the lifetime of a server session to 2 h.

But note: after 2 h the user is disconnected. This behavior shown by almost all servers to avoid resource shortages.

The message has been changed to express that (better as before, I hope).

rbudde, 2019-09-06:
tested

closed 2019-09-06 by rbudde