OpenRoberta/openroberta-lab #247: Calliope new dal available

bjost2s, 2019-09-05:
here https://github.com/calliope-mini/microbit-dal/releases/tag/v2.2
we should rebase on this

bjost2s, 2020-01-03:
Reviewed by the calliope team

bjost2s, 2020-01-05:
The attached program is one from the gallery (a little game). It is not compilable on the testserver anymore, `region RAM overflowed with stack collect`

[Diamantenjagd.zip](https://github.com/OpenRoberta/openroberta-lab/files/4023682/Diamantenjagd.zip)

rbudde, 2020-01-07:
this is a general problem of the code geenrator. Compiling a program with 9 variables for example, without any other behavior, crashes the linker.

Changed the code generator to fix this.

bjost2s, 2020-01-08:
Seems that the bluetooth stack uses too much memory. So one solution is to offer the calliope plugin with the bluetooth stack and without (with the radio stack) to make bigger programs compilable.

rbudde, 2020-01-10:
by adding a new noblu calliope plugin, the problem vanishes. It is likely, that it will re-appear, if either new dal version arrive or user write large programs.

See for a discussion related to microbit/gcc/newlib&locales the following resources:

`https://github.com/bbcmicrobit/micropython/issues/363` and the issue referred there: `https://bugs.launchpad.net/gcc-arm-embedded/+bug/1686067`

closed 2020-04-24 by bjost2s