OpenRoberta/openroberta-lab #249: compilation error on buster on a raspberry pi 3

corvus-albus, 2019-09-05:
I try to build a openroberta-server on a raspberry pi 3. I downloaded the latest raspbian image, whitch comes with debian buster.
I installed the the "default-jdk"-package. It gives
$ javac -version
javac 11.0.3
maven:  mvn -v
Apache Maven 3.6.0
and cloned openroberta-lab as suggested.

Compilation (mvn -e clean install) ends with an error message. You can find the error log attaches.
Should I go back to debian stretch or is there another solution?

Thanks in advance for any hint.

Cheers corvus-albus

[error_log_20190905.txt](https://github.com/OpenRoberta/openroberta-lab/files/3580404/error_log_20190905.txt)

VinArt, 2019-09-06:
You can also try java 8 if that is available for your distribution.

corvus-albus, 2019-09-06:
Thank you for your hint. I finally managed to install oracle java 8 manually and it compiled without errors.

closed 2019-09-06 by corvus-albus