OpenRoberta/openroberta-lab #250: "Help/Info" window does not close

Thorlogy, 2019-09-06:
**Describe the bug**
 - The help window does not close after switching the hardware system.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to OR Lab
2. Click on NXT
3. Switch to mBot
4. See error

**Expected behavior**
 - The help/info window should close 

**Screenshots**
[image]

**Device information**
 - PC
 - OS: Windows
 - Browser Chorme 

**Additional context**
- Try the same with the sim-window. This works fine.

bjost2s, 2020-01-29:
seems to be fixed in the meanwhile

boonto, 2020-01-29:
It still happens when switching to Bot'n Roll or BOB3, robots that do not have any associated help.