OpenRoberta/openroberta-lab #252: Sensebox does not get flashed with generated binary data

VinArt, 2019-09-09:
**Describe the bug**
 - The program that is delivered to the frontend is wrongly read-out and later encoded, so the resulting file is unsuitable to be flashed onto the sensebox. Sensebox consumes the file, but stays in the programming mode.

The generated binary should be read out as a byetarray and not as a string with encoding.

bjost2s, 2019-09-11:
@VinArt, there are changes also for other robots than sensebox, please expand the issue so that we do not forget to test everything!

VinArt, 2019-09-11:
test that eclipse's refactor->rename worked for following robots:
- [ ] mbot
- [ ] botnroll
- [ ] arduino

bjost2s, 2019-09-11:
+1 for a hotfix, non working plugins are the main reason for no users :-(

rbudde, 2019-09-13:
Upload to Sensework working.

closed 2019-09-13 by rbudde