OpenRoberta/openroberta-lab #259: Open Roberta does not load, if cookies are disabled in the browser

philippmaurer, 2019-09-27:
**Describe the bug**
- If you deactivate cookies in your browser, the browser will throw an error while still loading initially and the lab will not load. 
- As a direct response to that, the Open Roberta Lab will try to log that error to the server. This will be prohibited by the new session check, which allows only those calls from a client, that have previously sent a initial request to the server. Yet, because the lab does not load the initial request was never sent and the error log request is rejected. This is shown to the user via an "Your browser tab is invalid" error promt.

**To Reproduce**
Steps to reproduce the behavior:
1. Open your browser settings and go to the cookies tab. Deactivate cookies
2. Open a new tab and open the developer console in it (to see the error message later)
3. Open the Open Roberta Lab in that tab
4. See error

**Expected behavior**
 - The Open Roberta Lab should either load normally, even if cookies are disabled, and should simply hide all cookie related functionalities (This happens currently with the Interet Explorer)
 - Or the user should be prompted a helpful error message, that indicates the problem and helps him change his cookie settings in the browser

**Screenshots**
[image]

**Device information**
 - Desktop
 - Windows 10
 - FireFox 52.9.0, FireFox 60.9.0, Chrome 77.0.3865.90, Opera 63.0.3368.94, Edge 40.15063.674.0

**Additional context**
 - This Bug is NOT present with: Internet Explorer 11.1928.15063.0

boonto, 2019-12-30:
Should be solved with #38.

bjost2s, 2020-01-02:
still the same issue

bjost2s, 2020-01-03:
We use the sessionid to identify sessions, this is impossible if the user does not allow cookies at all. For this special case it is hard to find a workaround, because we cannot allow with every request a new session. Maybe we could use the url instead of cookies to store the sessionid, but this might be more work and cannot be solved till the next release.

Therefore we should now simply extend the message with a hint that enabling cookies for this site will solve the problem.

bjost2s, 2020-01-09:
together with this 9a5af389df3c3c7b458bf7f9e3f064cd071f8737, thanks to @rbudde seems to be resolved

closed 2020-01-10 by VinArt