OpenRoberta/openroberta-lab #260: simulation window gets all white if you click "EV3 - open the robots view" or "open the sensors' data view"

corvus-albus, 2019-09-27:
Choose ev3dev (or NXT/Calliope/...) and open the simulation window. Then click on "open the robots view" or "open the sensors' data view". The simulation window gets white you can't see any simulation.  

It was still working in ORS 3.5.2. I see the same behaviour on my self hosted ORServer. If I switch back to 3.5.2 anything works fine.
Cheers 
corvus-albus

corvus-albus, 2019-09-28:
I tried it again at Saturday evening (22:30) and it's working properly now.

bjost2s, 2019-10-01:
Maybe a problem of cached data? Please close, if it is not reproducable, thank you.

corvus-albus, 2019-10-01:
The popup was build outside the canvas. You could change zoom of the browser window  (Ctrl- -) and the popup appeared.
It is solved for me now.
Cheers corvus-albus

closed 2019-10-01 by corvus-albus