OpenRoberta/openroberta-lab #262: The program could not be generated. Pin 43 is incorrect in block LCD

sebastiankmilo, 2019-10-02:
**Describe the bug**
I can not upload my program to Arduino mega, it said that The program could not be generated. Pin 50 is incorrect in block LCD, but the Arduino mega has pin 50
**To Reproduce**
Steps to reproduce the behavior:
1. Go to galery
2. find efi
3. click on efi
4. connect arduino mega
5. upload de program

**Expected behavior**
 - upload de program 

**Device information**
 - Type [Desktop]
 - OS: [windows 10
 - Browser chrome and Firefox

**Additional context**
 - a 2 month ago, i can upload de same program

bjost2s, 2019-10-04:
@VinArt , could you please check if the (new) checker for the pins is working correct? The program and configuration seems to be valid!

bjost2s, 2019-10-04:
@VinArt, there is already a hotfix branch for issue #258, please commit on this, if there is a solution.

boonto, 2019-10-04:
I can take a look and maybe fix it.

bjost2s, 2019-11-08:
works now

closed 2019-11-08 by bjost2s