OpenRoberta/openroberta-lab #266: reset EV3 c4ev3 gyroscope does not work in simulation

tcpan, 2019-10-08:
Resetting EV3 c4ev3 gyroscope in simulation does not appear to work.  All settings are default.  Example code: 

#define PROGRAM_NAME "NEPOprog"
#define WHEEL_DIAMETER 5.6
#define TRACK_WIDTH 18.0

#define PROGRAM_NAME "NEPOprog"
#define WHEEL_DIAMETER 5.6
#define TRACK_WIDTH 18.0

#include <ev3.h>
#include <math.h>
#include <list>
#include "NEPODefs.h"

double ___angle = 0;

int main () {
    NEPOInitEV3();
    NEPOSetAllSensors(NULL, EV3Gyro, NULL, NULL);
    NEPOResetEV3GyroSensor(IN_2);
    
    
    for (float ___i = 0; ___i < 4; ___i += 1) {
        RotateMotorForAngle(OUT_BC, Speed(30), (20 * 360) / (M_PI * WHEEL_DIAMETER));
        NEPOResetEV3GyroSensor(IN_2);
        while ( ReadEV3GyroSensor(IN_2, EV3GyroAngle) < 90 ) {
            RotateMotorForAngleWithTurn(OUT_BC, Speed(30), (1 * TRACK_WIDTH / WHEEL_DIAMETER), -200);
        }
    }
    
    NEPOFreeEV3();
    return 0;
}

simonedegiacomi, 2019-10-08:
Hi @tcpan 
What wrong behavior are you observing in the simulation? and what do you expect instead?
Could you please also post a screenshot of the NEPO program?

I tried to run in the simulator the following program:
[image]

and, while the robot rotates, I can see in the screen the gyro angle going up and then starting again from zero after some time, as expected.

tcpan, 2019-10-09:
@Simone,

Thanks for getting back to me.

It's the attached program:

Problem:  The gyroscope is not resetting. The program moves the robot 20
cm, makes 1 turn,  then just kept going straight for the next 3 x 20cm
drives.  The angles printed out are
0, 9x.xxxx
9x.xxxx, 9x.xxxx
9x.xxxx, 9x.xxxx

and so on.  the gyro value does not change after it reached above 90
degrees.

Expected:  the gyroscope should reset to 0, so any print out should read
0 9x.xxx,
0, 9y.yyyy
and so on.

By the way, I tried your program, but I think the gyro is not resetting in
your program either.  It looks like the value goes down is due to the angle
readout wrapping around 360 (mod 360).  If you set your inner loop to
repeat just 1 time, you'll see the read out increasing indicating that
resetting is not happening.  the expected behavior would be the value
staying at around 20.

I am using EV3 c4ev3, testing in MS edge, firefox, and chrome browsers.

Tony

simonedegiacomi, 2019-10-16:
You're right, the gyro reset is [currently not implemented](https://github.com/OpenRoberta/openroberta-lab/blob/457357bf63c62099481efbfb20cf1baee4362141/OpenRobertaServer/staticResources/js/app/wedoInterpreter/interpreter.robotMbedBehaviour.js#L330).

tcpan, 2019-10-16:
Is it possible to implement it quickly?  I am using the tool to help with programming and debugging ev3s for an elementary school LEGO FIRST competition.   It would be good to be able to continue using it.

I’ll test this but do you happen to know if the gyro reset works when downloaded to the ev3 brick?

simonedegiacomi, 2019-10-17:
@tcpan the gyro reset should work on the real EV3.
I've now implemented the reset in the simulator, it should be available in the next release of the lab

boonto, 2019-10-28:
Just checked, works as intended. Thank you for fixing this Simone.

rbudde, 2020-05-07:
The fix doesn't work correct, because the port is not used. Furthermore the fix was applied to a js file instead to a ts file. Thus is was overwritten with the next change.

This is a PROBLEM: some javascript files are generated, most not. The whole simulator is generated, for example.

bjost2s, 2020-05-07:
works fine, tested with single and multiple simulation

rbudde, 2020-05-08:
in the case (this is NOT very reasonable), that a ev3 has more than 1 gyro, only the last one used gets resetted correct. The attached program (xml->txt because github disallows xml) 
[ev3-2gyros-test-turn.txt](https://github.com/OpenRoberta/openroberta-lab/files/4598607/ev3-2gyros-test-turn.txt)
should show for both gyros about 20°

rbudde, 2020-05-11:
Test program passes. Error corrected.

closed 2020-05-12 by bjost2s