OpenRoberta/openroberta-lab #267: Differences in performance in c4ev3

munka21, 2019-10-08:
**Describe the bug**
There are large differences in performance between driving blocks. So if I want to set the power of both engines in a loop as often as possible, and if I use a single drive block for both engines, a loop will have about 44,000 runs. But if I use two single blocks (for each engine separate block) only 220 runs per second are done in the loop (Under Screenshots).

**Expected behavior**
You get the same or similar performance in both cases

**Screenshots**
Case Where you set two engines at the same time
[image]

Case Where You set Each Motor Separately
[image]

**Device information**
Windows-PC
Chrome/Firefox

simonedegiacomi, 2019-10-08:
As I anticipated in the Gitter channel, I added to c4ev3 an explicit delay to prevent the program to write too frequently to the pwm device (which controls the motor). Without the delay I observed an inconsistent behavior when setting the speed of the motor inside a loop.

Reading the code I found out that I added the delay only if the motor is controlled singularly. This explains the difference on performance.

@munka21 Currently I don't have an EV3 to try, but did you find some inconsistencies using the first program (which controls the motors with only one block)? Also using high speeds (like >80%)?

boonto, 2019-11-12:
Hey @simonedegiacomi how far are you with this?

simonedegiacomi, 2019-11-14:
The problem should now be solved, but I didn't try it myself (I asked @peehrlich2 that was very kind to test some programs). On Sunday I should be able to try some programs on a EV3

boonto, 2019-11-15:
Nice, I can test today.

boonto, 2019-11-15:
I just tested those two programs:
In the Lab I get: `45617` and `220`
On your branch I get: `1` and `189503`
Single motors are definitely faster now but something changed with the drive block.

simonedegiacomi, 2019-11-19:
Now the bug should be fixed (without using caches or waits)

boonto, 2019-11-20:
Values now: `56857` and `20035` so both are fast now, so I guess this is resolved.

How come there is a magnitude difference from the previous fix for the second value?

simonedegiacomi, 2019-11-21:
> How come there is a magnitude difference from the previous fix for the second value?

Because in the previous "fix", the function to control the motor was caching the previous set speed. If the speed was the same, it didn't call the function that actually wrote to the motor device. With the new fix there is no such cache, it always writes to the motor device. This includes I/O operations, so it slower.

Anyway I think that it's better to have it like this because it's less bug-prone and the speed should be enough for almost any problem (including PID and similar). What do yo think? does this make sense?

boonto, 2019-11-21:
That sounds reasonable and definitely more stable.

closed 2019-12-19 by boonto