OpenRoberta/openroberta-lab #273: Background upload doesn't work with Chrome, but with Firefox

rbudde, 2019-10-21:
**Describe the bug**
Press the upload background image button in the simulation. The image is visible in Firefox, but not in Chrome

**Expected behavior**
 should work with ALL browsers

**Device information**
 Windows 8, Chrome Version 77.0.3865.120 (Offizieller Build) (64-Bit)

boonto, 2019-11-14:
After uploading in Firefox the simulation is fully zoomed in and if I zoom out I get the Error `NS_ERROR_FAILURE` in the drawBackground method in scene.js line 76.

boonto, 2019-11-14:
In Chromium on Linux the Image is uploaded, but not autoselected. I have to cycle through the images to get to the new one.

boonto, 2019-11-14:
But only sometimes... similar in Chrome, here the image is not uploaded most of the time.
For some reason the change event is (sometimes) not triggered.

A working solution is to use a static input element in the index.html instead of dynamically creating one every time.

VinArt, 2019-12-20:
Can upload now, although any file type.

closed 2019-12-20 by VinArt