OpenRoberta/openroberta-lab #274: Multiple small issues with Sensebox

philippmaurer, 2019-10-22:
**Describe the bugs**
 - If you add the WIFI actuator to your robot configuration, the program will compile, but it will not start on the sensebox. Using the sensemap, or connecting to a WiFi, is therefore impossible.
 - The BMP280 sensor outputs wrong data in 30% of the cases. The data are sometimes 5 times higher, sometimes 20% off the real values. You either get the right or the wrong data during one whole program execution. I was not able to identify a way to reproduce that behavior consistently. It is a bug though, because it does not occure at all, if you use the SenseBox-Blockly editor.
 - The OLED display is built visibly slower in the Open Roberta Lab, than if you build the same markup using the SenseBox-Blockly editor. For example, if you enter 8 lines of text, the text is show immedeately using the SenseBox-Blockly editor, but takes about 1.5 seconds to built using the Open Roberta Lab.
 - It is not possible to enter spaces in WIFI SSIDs, eventhough its an often found character nowadays. Some special characters, like the dollar sign ("$"), are also forbidden for the password for no obvious reason.

**Additional context**
 - An Sensebox EDU was used for testing
 - More information can be given by @VinArt

VinArt, 2019-10-23:
1, 2 and 4 should be now fixed with the hotfix, 3 will be tackled later, since it requires (most probably) the update of the resource container, so this will not be hotfixed.

bjost2s, 2019-11-14:
Seems that this issue is already fixed and released, but never reviewed.
Will be done with the next release.

closed 2020-01-03 by bjost2s