OpenRoberta/openroberta-lab #275: Can not open simulation if Norton Internet Security is running

philippmaurer, 2019-10-23:
**Describe the bug**
It is not possible to open the simulation if Nortin Internet Security (NIS) is running on the device.

**To Reproduce**
Steps to reproduce the behavior:
1. Use a PC with NIS
2. Go to the ORLab and select a robot with a simulation
3. Press the "SIM" button
4. See error / Nothing happens

**Expected behavior**
 - The simulation should be usable, even if NIS is installed.

**Screenshots**
[image]

**Device information**
 - Windows 10 PC
 - FireFox

**Additional information**
 - Calliope Simulation

VinArt, 2019-10-24:
So... Is it possible to add an exception in NIS to allow it to run the simulation?...

philippmaurer, 2019-10-24:
The user did not mention a specific error message that was shown in Norton Internet Security, so even if it is possible, the user might not be able to do it.
The javascript error (That the user does not allow the browser to access the microphone) itself should still be catchable i think.