OpenRoberta/openroberta-lab #278: Replace usages of BlocklyMethods for C robots

boonto, 2019-10-24:
Continuation issue to partially solved issues #95 and #230:
`BlocklyMethods` and `NEPODefs` still have to be removed from Java and C based robots.

- [x] Find all "BlocklyMethods.py" / "NEPOdefs.h" etc.
- [x] Find common C methods
- [ ] Extract C one liners (constants?) directly into code generation
  - [ ] Arduino based
  - [ ] c4ev3
  - [ ] Calliope
  - [ ] NXT
- [ ] Combine C multi liners into common helper methods
  - [ ] Arduino based
  - [ ] c4ev3
  - [ ] Calliope
  - [ ] NXT
- [ ] Implement class in all C robots
  - [ ] Arduino based
  - [ ] c4ev3
  - [ ] Calliope
  - [ ] NXT
- [ ] Default implementations for C
- [ ] Special implementations for C robots that don't support default
- [ ] Unit tests
  - [ ] Arduino based
  - [ ] c4ev3
  - [ ] Calliope
  - [ ] NXT
- [ ] Test each method on robots
  - [ ] Arduino based
  - [ ] c4ev3
  - [ ] Calliope
  - [ ] NXT

boonto, 2020-02-12:
We discussed some changes today.
The original goal was to simplify and show the code for everything from math to lists.
Now, for C specifically, we decided to create helper methods for math operations but omit the list operations (together with HAL-like abstractions, which should be moved into its own file at some point) in the `NEPODefs.h`. As these methods are mostly iterator based there is not a lot of didactic value to be gained.

boonto, 2020-02-13:
I also changed the math constants across all C robots to follow the `M_` naming.

rbudde, 2020-03-19:
hard to test manually.

closed 2020-03-19 by rbudde