OpenRoberta/openroberta-lab #279: Can not load the NEPO4Arduino background image

philippmaurer, 2019-10-24:
**Describe the bug**
The NEPO4Arduino background image can not be loaded.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the Open Roberta Lab and select Uno, Nano or Mega as robot
4. See error (White background)

**Expected behavior**
 - A transparent image of an arduino in the programming background

**Screenshots**
[image]

**Device information**
 - Laptop with Windows 10
 - FireFox 60.9 esr

bjost2s, 2019-11-03:
This is because we have not had  this background pictures yet. They could be generated from the pictures for the startup popup, nano.png, uno.png and mego.png located here https://github.com/OpenRoberta/openroberta-lab/tree/develop/OpenRobertaServer/staticResources/css/img

Ramanjs, 2019-12-17:
please see #374

boonto, 2019-12-17:
The background image is currently being created.

Ramanjs, 2019-12-18:
please see #382

boonto, 2019-12-23:
Resolved by GCI student.

closed 2019-12-23 by boonto