OpenRoberta/openroberta-lab #280: [newcomers only] Make crosscompiler executables names configurable

VinArt, 2019-10-24:
Problem:

OpenRoberta has many Robot plugins, most of which require compilation of the generated source code to binary.
Currently crosscompiler names are hardcoded in Java classes, examples can be found in C4Ev3SourceCompiler class, line 30.
This sometimes leads to a problem that system comiler names do not match and the compilation cannot succeed. For the supported
linux system (ubuntu 18.04) this is not the case, but could be for other systems, where the compiler is called differently.
Related: #128 

Proposed solution:

it would be better to have these names configurable through .property files, for example ev3c4ev3.properties, as the values 
specified there can be overwritten with the -d flag when starting the server, like robot.plugin.compiler.executable = name_goes_here.
In Eclipse this is done in the run configuration under arguments -> program arguments.

Possible steps to resolve the issue:

- extend CompilerSetupBean class to contain String compilerBinaryName field
- extend CompilerSetupBean.Builder class to contain a method to set the said field
- add a line to CompilerSetupWorker class in execute method to use the implemented methods to put the needed data into CompilerSetupBean
- use this information in *CompilerWorker classes to get needed information and pass it further to be used as the compiler binary name, for example to C4Ev3SourceCompiler class
- CompilerSetupWorker has access to pluginProperties, which means that a coresponding line, like robot.plugin.compiler.executable = name_goes_here needs to be added to the .properties files (in src/main/resources)
- this could be easily tested, since supplying the wrong compiler name would cause an exception during the compilation attempt (after the play button press)

ShubhiAgrawal2000, 2020-03-15:
Hii I'm a newbie and want to contribute to this. Please guide me for the same.

VasilisPoulos, 2020-03-16:
Hi, i changed C4Ev3SourceCompiler to get the compiler name from the .properies file ,tested it on the http://localhost:1999/ server and it seems to be working fine. I have some questions regarding the changes i made!

boonto, 2020-03-16:
@ShubhiAgrawal2000 try following the "Possible steps" after cloning and running the server.

@VasilisPoulos nice, what are your questions?

closed 2021-02-05 by bjost2s