OpenRoberta/openroberta-lab #282: Support for Makeblock Me LED Matrix 8 X 16

bjost2s, 2019-10-29:
**Is your feature request related to a problem? Please describe.**
 - Currently it is not possible to program this LED Matrix, but a new Mbot comes with it

**Describe the solution you'd like**
 - A block, similar to the LED block for the calliope display should be integrated

**Additional context**
 - I think this is already partially integrated

boonto, 2020-01-13:
Using "invert" with a image variable works but it doesn't work when entering a new image directly.

Another note: shifting images out of the display and back in again removes the data that was "outside" of the image:
[image]
But that is expected unless we want to keep a copy of each image around at all times.

boonto, 2020-01-13:
Invert works with the latest resources and the second note was not really a problem.

closed 2020-01-13 by boonto