OpenRoberta/openroberta-lab #284: Source code editor

VinArt, 2019-10-31:
Currently it is not possible to edit source code in the source code view pane, this brings disadvantages for power users as well as for testers.

In order to tackle this problem an approach was devised that includes the following steps:
- Introduce a menu entry under "Program" menu - "Source (Code?) Editor".
- Create a new pane alike to logging or gallery to block out user interaction with blockly.
- Said pane needs to have a back button, as all others, and a set of five buttons: import code from blockly workspace, download source, upload source, build source and run on the robot.
- The build button should be deactivated for robots that do not require the build - NAO, ev3dev, WeDo
- The editor should replace tabs with space on copy-paste and on tab press.
- Contents of the editor will be preserved for current user session with the possibility for the user to replace current edited code with the one generated from blockly via the import button.

Pre-requisites to this issue were workflows #114 #180 and the 199eeecb9cfbad055b6a372c510072551505a081

It may happen so that new workflows will be introduced, for example for the build function.
#149 should be considered as an extension/related of this issue.

things to consider:
- Should the server distinguish between user provided code and generated code?
- Should user-edited code be saved and not only exported?
- Should compiler errors be delivered via a pop-up or in a separate pane?

possible extensions:
- A button to download lab-specific files (headers)

VinArt, 2019-11-07:
As of now source code editor is implemented as a text area. However there are possibilities to utilise a better editor with extensive set of features (line numbering, code folding, highlighting, syntax checking etc.) but this comes at a cost of the size of transmitted JS. 

The following was tested:
- codeflask+prism was impossible to properly style, has least features, but the size is ~30kB
- ace editor - works fine, size ~400kB
- codemirror - requires additional styling, size ~400kB

there are other editors, but in the end a feature-complete editor has the size ~300-400kB. I personally think that the best option would be to use Ace as it requires the least amount of work and is feature-rich. Together with Ace I propose minimisation of our JS which will give us back ~350kB + ~100kB for interpreter and with removal of Prettify (15-20kB) we will be at a similar or better overall transmitted byte count.

VinArt, 2019-11-20:
some requirements:

Feature: source code editor

An editor is needed, where users can modify the generated from the blockly blocks code, execute it on the robot, save and import as well as check that it could be built.

1. The editor must have a possibility to import the generated code. Current blockly workspace state needs to be the reference.
2. The editor must have a possibility to export (allow to download) the current state of the edited code. If the code was modified by the user, the modified version needs to be saved.
3. The editor must have a possibility to import source code effectively replacing its contents with the contents of the uploaded file. For each specific robot systems only corresponding file extensions must be allowed. E.g. .java for ev3 lejos, .py for NAO, etc.
4. The editor must have a possibility to build the code that is currently it's contents and provide as an output the output of the compiler as a popup to the user in case there were build errors, in case there were no errors and it is possible to compile current code a popup needs to be presented to the user with the confirmation of the success of the compilation.
5. For systems that do not have an explicit compilation step (e.g. python based robots) build button needs to be deactivated.
6. The editor must have a possibility to build and then execute the code on the robot.
6a. When the robot is connected by any means (token/agent/auto) the run button needs to be activated and after pressing said button the program will be executed on the robot for all robots except the ones that have auto connection. In this case a popup with a download option must be presented to the user.
6b. When the robot is disconnected the run button must be deactivated.
7. The editor must have line numbering on the left of the editor.
8. Code lines must not be wrapped and horisontal scrolling must be provided instead when the length of the line is more than the width of the editor.
9. Vertical scrolling must be provided when the contents of the editor does not fit the height of the window.
10. Editor backgorund must be white (#FFFFFF)
11. Background of the editor pane must be gray (#EEEEEE)
12. Background of the line numbers must be gray (#EEEEEE)
13. Base text color must be black (#000000)
14. Code needs to be highlighted according to the language. Python, java and c-like languages need to be supported.
15. The menu entry to open the source code editor must be in the edit menu under the "open/close source code view" and should be called "open source code editor"
16. The editor must be initialised with the source code when the user selects the menu entry to open the source code editor.
17. Button to return to the previous view must take the user back to the blockly workspace.
18. User must be notified, that when leaving the editor if there were changes to the source code that these changes will be lost.
18a. User must be notified when switching the robot system.
18b. User must be notified when pressing the back button to go back to the blockly workspace.
19. Edit menu must be disabled while in the source code editor view.
20. The library that does pretty-print for source code view needs to be replaced with the library that implements the editor.
20a. Source code view must not allow the user to modify its contents.
20b. Source code view requires scrolling when the source code does not fit the dimensions of the source code view pane.

VinArt, 2020-01-02:
A commit will be made to display source code in the source code view 12px lower to make it more readable

bjost2s, 2020-01-04:
This feature is a big enhancement and we will release it "as it is" to let the community test it so that we can use the feedback for further improvements in future releases.

closed 2020-01-04 by bjost2s