OpenRoberta/openroberta-lab #288: Memory leak on uploading (and zooming) background image with Firefox ESR 68.2.0 64 bit

germanicianus, 2019-11-07:
The first time I uploaded a simulation background image, it has been displayed, but was extremely zoomed in. Therefore I tried to zoom out with the mouse wheel (rolling it back). This lead to Firefox eating up all the RAM and thrashing the paging file. In the end even the mouse pointer was frozen and I had to hard reset the computer.

System:
Windows 7 SP 1 on latest patch level
Firefox ESR 68.2.0 64 bit

Steps to reproduce:
1. Open https://lab.open-roberta.org/ and choose EV3.
1. Click SIM (right side).
1. Right click simulation background between back of the robot and black track, click "Save Image as..." and save it somewhere.
1. Click background image upload button and select previously saved image.
1. Image is displayed extremely large.
1. Open task manager with Ctrl-Shift-Esc and see Firefox eating up RAM slowly.
1. Try to zoom out with mouse wheel rolling it back one or two clicks (not to much).
1. See Firefox eating up RAM faster.

Once I had Firefox displaying the just uploaded background just briefly (also extremely large) - see the screenshot at the end.

Workaround:
After uploading the background image, reload the page, select EV3 again and cycle through the background. The just uploaded one is there and will be displayed correctly.

What I also miss is the documentation on which pixel size the image has to have for which virtual cm size. I found that out uploading an image and comparing the pixel size with the ultrasonic sensor reading - it's apparently pixel/3 = cm.

[image]

boonto, 2019-11-15:
I split off the zoom problem in #318. This should fix the issue of uncontrollably rising memory consumption.

Additionally, I compared memory consumption of Firefox and Chrome when uploading a background image.
For Chrome the memory consumption remained similar after uploading multiple backgrounds.
For Firefox uploading an image has a definitive impact on the memory consumption. By uploading multiple images in a row the memory cost of the Lab tab alone can eat up to 700 MB in my tests. After GC ticks in the consumption lowers again.

rishabhdeepsingh, 2019-11-18:
This problem is on Chrome too and I am not even able to upload the image.

boonto, 2019-11-19:
@rishabhdeepsingh the uploading should be fixed with #273. Do you get significantly higher memory consumption on Chrome? I didnt with Chrome & Chromium on Ubuntu 18.04.

rishabhdeepsingh, 2019-11-19:
No, I am not getting any high memory consumption.

boonto, 2019-11-19:
Okay, then #273 should be enough for Chrome :)

closed 2023-08-21 by bjost2s