OpenRoberta/openroberta-lab #289: ArduinoBrickValidatorVisitor needs revision

VinArt, 2019-11-08:
ArduinoBrickValidatorVisitor has a method override  for checkSensorPort. This is however more fully implemented in the super class, so it becomes unclear why this method is overridden. Another thing is that for the sensors mentioned in the method (INFRARED_SENSING and GYRO_SENSING) there seems to be no way to hit their respective case statement since configuration blocks for them would be either null when not used and will have different component type when used.

1. Check that sensors from ArduinoBrickValidatorVisitor and AbstractBrickValidatorVisitor  with their respective configuration blocks are not hitting the case clauses in switch ( usedConfigurationBlock.getComponentType() ) 
2. Check whether configuration blocks at all can have types mentioned in checkSensorPort method
3. Remove dead code if needed.

Ramanjs, 2020-01-05:
please see #503

boonto, 2020-01-14:
Resolved by GCI student.

closed 2020-01-14 by boonto