OpenRoberta/openroberta-lab #293: You can not see the users, you have shared your program with, in the program list

philippmaurer, 2019-11-08:
**Describe the bug**
In the "my programs" list is a column that shows the users you have shared a program with. Only the last user you have shared a program with is shown there and a dropdown icon, which indicates that you can see the rest if you click on it. 
If you click on the dropdown icon, the list pop ups, but the "my programs" view gets closed simultaneously and the program loads, as if you wold have clicked somewhere else on the list entry.

**To Reproduce**
Steps to reproduce the behavior:
1. Log into an account in the Open Roberta Lab, which has at least one saved program.
2. Open the "my programs" view and share your program with at least two users.
3. Go back to the "my programs" view and try to click on the  dropdown icon near the name of the last person you have shared your program with.
4. See error

**Expected behavior**
The list of the users you shared your program with should expand and the program should not be loaded into your program pane.

**Screenshots**
[image]

**Device information**
 - Desktop
 - Windows 10
 - FireFox 60.9

boonto, 2020-01-15:
Resolved by a GCI student in 25afe178491c48ff48ca5abdbe3b900b17583e49.

closed 2020-01-15 by boonto

KevinL10, 2020-01-20:
The dropdown will now show the users without closing, but there is an issue if you have more than one program. If you toggle only one of the user dropdowns, all of the black "collapse triangles" will be toggled, as shown in this attachment. The main lab does not exhibit this behaviour.
[image]