OpenRoberta/openroberta-lab #302: Integrate Festo Bionics4Education

boonto, 2019-11-12:
(Bionics4Education)[https://www.bionics4education.com/web/lang/en/1_education.html] by Festo is a robot set based on ESP32, the set includes parts to build 3 different models (Elephant trunk, fish and chameleon tongue).
4 Motors can be used to actuate the flexible grippers/fins.
We want to integrate this set into the Lab, it will be in the subgroup of Arduinos.

- [x] Check capabilities
- [x] Add project
- [ ] Adapt compiler workers
- [ ] Add transfer/detection capabilities to (Connector)[https://github.com/OpenRoberta/openroberta-connector]

boonto, 2019-11-22:
First integration is done, works on Ubuntu 18.04 with Connector on `develop` branch.
To install xtensa compiler follow [this](https://gitdemo.readthedocs.io/en/latest/linux-setup.html).
I did these steps on my machine:
```
sudo apt-get install git wget make libncurses-dev flex bison gperf python python-serial
cd ~
wget https://dl.espressif.com/dl/xtensa-esp32-elf-linux64-1.22.0-61-gab8375a-5.2.0.tar.gz
mkdir -p ~/esp
cd ~/esp
tar -xzf ~/xtensa-esp32-elf-linux64-1.22.0-61-gab8375a-5.2.0.tar.gz
rm ~/xtensa-esp32-elf-linux64-1.22.0-61-gab8375a-5.2.0.tar.gz
echo "export PATH=\$PATH:\$HOME/esp/xtensa-esp32-elf/bin" >> ~/.profile
```

boonto, 2019-11-29:
Currently the servo block contains reference to the "SG90" servo from Arduino, this should be removed.

boonto, 2019-11-29:
Was handled in 772c58e375e3f4a53ebb0d3cbd6a765844ce1f23.

boonto, 2020-01-29:
To build the necessary compiler for Raspberry Pi I was able to follow the "Alternative Step 1" instructions [here](https://gitdemo.readthedocs.io/en/latest/linux-setup.html).
However, `libtool` is necessary as well which can be installed using `sudo apt-get install libtool-bin`.

Afterwards, I copied the built compiler to the home directory and added it to the path, similar to before.

boonto, 2020-01-30:
Transfer on a Windows 10 device works as well.
Requirements are a Python installation and the `pyserial` module.
I installed it in an admin powershell using `python -m pip install pyserial`.

We will have to think of a way to simplify this for end users.

boonto, 2020-03-24:
I improved/simplified the Python requirement and packing of esptool by automatically checking for Python install via PATH and registry (on Windows) and installing it on demand. This way less resources are spread across all projects. With the same background, everything necessary will be sent in the packed zip file, instead of adding it to the resources.

bjost2s, 2020-06-02:
will be published soon with beta label, for remaining bugs please create a separate issue

closed 2020-06-02 by bjost2s