OpenRoberta/openroberta-lab #314: Send data to opensense map block should break the workflow when wifi configuration block is not used

VinArt, 2019-11-13:
**Describe the bug**
 - if the send block is used without the configuration block then the porgram can be compiled, but it shows the wrong error, and namely the compiler output

**Expected behavior**
 -the program should not be compiled and the workflow should be interrupted before at configuration validation stage

boonto, 2019-12-19:
It breaks the workflow, however the message is confusing: "This actuator is not set to the port!" does not make sense for the WIFI/SD card.

VinArt, 2020-01-03:
@boonto it comes from the blocks being under "Action" category and the error message is CONFIGURATION_ERROR_ACTOR_MISSING. We can provide specific messages for such cases. SenseboxBrickValidatorVisitor -> line 75. New issue?

bjost2s, 2020-01-03:
No need for a new issue, I think. Just fix this one without producing a new one.

bjost2s, 2020-01-07:
Now we have two new messages, different from the old one only in the subject:

This actuator is not configured. Please add the corresponding block in the configuration tab!
SD card is not configured. Please add the corresponding block in the configuration tab!
This sensor is not configured. Please add the corresponding block in the configuration tab!
WiFi is not configured. Please add the corresponding block in the configuration tab!"

Wouldn't it be possible to send back the name of the sensor/actuator (available in the checker) to avoid redundancy in the messages?
e.g. "$ is not configured. Please add the corresponding block in the configuration tab!"

rbudde, 2020-01-07:
What substitutions are done for $? "This actuator", "SD card". Substitutions cannot be internationalized easiliy.

bjost2s, 2020-01-07:
In best case only keys, like SENSOR_INFRARED, SENSOR_ULTRASONIC, SERVO_MOTOR, derived from blockly messages.

VinArt, 2020-01-08:
1. When we check blocks and put annotations on them we add `new NepoInfo.error(severity, message)`, where message is a `String`, so we would have to either invent some message format that would take the message key and the key for substitution or instead of only message supply a map, that the FE would the process. This would require changes in BE, FE and either *all* classes that ever annotate blocks with NepoInfo or create a method that would build the map and change the existing one to create a single-entry map.

2. I think this is not related to the breaking of the workflow when the configuration block is used.

boonto, 2020-01-10:
The SD card message is the same as the default actuator one. The openSenseMap mode has a specialised message with WIFI included.

closed 2020-01-10 by boonto