OpenRoberta/openroberta-lab #321: Unable to read No matching record found

rishabhdeepsingh, 2019-11-18:
**Describe the bug**
 - In the tutorials page, the text is the same colour as the background which should be black in colour.

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to the [lab](https://lab.open-roberta.org/#)
2. Click on 'tutorials'
3. No matching records found the text is not visible at all.

**Expected behaviour**
 - The colour should be black.

**Screenshots**
 
[image]

**Device information**
 - Type Desktop
 - OS: MacOs Catalina Version 10.15.1
 - Browser chrome Version 78.0.3904.97 (Official Build) (64-bit)

ArcSlayer121, 2019-12-05:
So I have found a way to solve this issue. We can change the background color of the table data by adding a style attribute in the class "no -records- found" and change the background color of the data.Kindly find the attatched screenshot of the same.We can do this by tr.no-records-found{ background-color : Black;
[image]

rishabhdeepsingh, 2019-12-05:
@ArcSlayer121  Would you like to work on this one?
You can submit a PR for the same.

nishanth1232, 2019-12-06:
Found a simple fix.
We can change the element concerned with the title to:
`<td colspan="9" style="color:black;">No matching records found</td>`
It should look like this:
[image]

nishanth1232, 2019-12-07:
@rishabhdeepsingh 
I claimed a task based on this issue in Google Code-In and opened a pull request for a fix.
Link - "https://github.com/OpenRoberta/openroberta-lab/pull/345/files"

boonto, 2019-12-09:
Resolved by a GCI student.

rishabhdeepsingh, 2019-12-09:
@nishanth1232  Really great. Hope to get more issues resolved soon.

boonto, 2019-12-17:
The text is readable now.

closed 2019-12-17 by boonto