OpenRoberta/openroberta-lab #323: Wait bluetooth connection block causes "server error" in simulation

simonedegiacomi, 2019-11-19:
**Describe the bug**
It's not possible to open the simulator if a program contains the "wait connection" block. The error that appears says that there is a server error.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new program
2. Create a Connection variable
3. Assign to the Connection variable the result of the wait connection block
4. Open the simulator

**Expected behavior**
If the simulator can simulate a bluetooth connection, then the simulator should open. Otherwise the error should provide more details to the users (ex: remove the bluetooth related blocks from the program).

Related to #224 

**Screenshots**
[image]

**Device information**
 - Type: Desktop
 - OS: Ubuntu 19.10
 - Browser: Firefox 70.0.1

boonto, 2019-12-10:
Currently in progress by a GCI student.

boonto, 2019-12-16:
Resolved by a GCI student.

closed 2019-12-17 by boonto