OpenRoberta/openroberta-lab #327: Adding the "Say" block first causes a "server error"

ensonic, 2019-11-22:
**Describe the bug**
 - A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Select ev3/ev3dev
2. Add a `Say` block
3. Click on `<>` to see the code
4. A popup "Attention Server-Error" appears

If I add a different block first (e.g. Drive forward), it works.

**Expected behavior**
 - A clear and concise description of what you expected to happen.

**Screenshots**
causes "server error"
[image]
works:
[image]

**Device information**
 - Type Desktop
 - OS: debian linux
 - Browser: chrome 78.0

boonto, 2019-12-06:
After some investigation this happens because I removed the `isSayTextUsed()` and replaced that by adding a dummy `VOICE` actor to the used actors when the say block is used. Then, when creating the used actor map no corresponding configuration component exists.

I think the real solution would be to add an internal configuration component for the voice capability, however this would only be easy for the new configuration.

For now there are two workarounds:
- go back to `isSayTextUsed()`
- check for the `VOICE` actor during map creation

I think the second option is best for now as it can be removed once we switch to the new configuration.

VinArt, 2019-12-20:
Works with say block as first block, though it might need a corresponding config block in future (e.g. language setting)

closed 2019-12-20 by VinArt