OpenRoberta/openroberta-lab #328: The text to speech in the simulation seems to always pronounciate in english

ensonic, 2019-11-22:
**Describe the bug**
When doing a programs for an ev3 robot and running this in the simulator the pronounciation sounds like english.

The relevant code is here:
https://github.com/OpenRoberta/openroberta-lab/blob/424dab9aa12c2959c6ee9e65a96b8d22ca91c489/OpenRobertaServer/staticResources/js/app/simulation/simulationLogic/robot.ev3.js#L158

When configuring the utterance in 
https://github.com/OpenRoberta/openroberta-lab/blob/424dab9aa12c2959c6ee9e65a96b8d22ca91c489/OpenRobertaServer/staticResources/js/app/simulation/simulationLogic/robot.ev3.js#L171

we might want to set `utterThis.lang` as well. See
https://developers.google.com/web/updates/2014/01/Web-apps-that-talk-Introduction-to-the-Speech-Synthesis-API

boonto, 2019-11-25:
Its actually already implemented implicitly through the voices. You can access it with the "setLanguage" block. Currently the default language is always English.

boonto, 2019-11-25:
It should however use browser or lab language by default, I guess.

bjost2s, 2019-12-06:
In best case it behaves like the real robot, so afaik it should have the browser's language as default (or English, if the browser's language is not supported).

VinArt, 2019-12-20:
works with both lab language and set language block

closed 2019-12-20 by VinArt