OpenRoberta/openroberta-lab #332: Integrate Arduino Uno Wifi

boonto, 2019-11-26:
As the Wifi edition of Arduino seems not to work with our current Uno plugin we have to adapt a specific one for the Wifi edition.

- [ ] Add as Arduino group plugin
- [ ] Adapt CompilerSetupWorker
- [ ] Add to Connector

boonto, 2019-12-02:
Similar to the festobionic implementation a different compiler is needed.
As the atmega4809 specs and libs are not part of the default `avr-gcc` implementation of Ubuntu, we need the modified one from Arduino. A link to the compiler can be found [here](https://github.com/arduino/Arduino/blob/master/hardware/package_index_bundled.json#L116) in the ArduinoIDE packages.
With these commands I added it to my path:
```
cd ~
wget http://downloads.arduino.cc/tools/avr-gcc-7.3.0-atmel3.6.1-arduino5-x86_64-pc-linux-gnu.tar.bz2
mkdir -p ~/avr-gcc
cd ~/avr-gcc
tar -xf ~/avr-gcc-7.3.0-atmel3.6.1-arduino5-x86_64-pc-linux-gnu.tar.bz2
rm ~/avr-gcc-7.3.0-atmel3.6.1-arduino5-x86_64-pc-linux-gnu.tar.bz2
echo "export PATH=\$PATH:\$HOME/avr-gcc/avr/bin" >> ~/.profile
```
Next, I uninstalled our recommended `avr-gcc` to use the custom one instead.

I also added functionality to the `robot.plugin.compiler.linux.dir` property which can be used to have both compilers running at the same time.
However, all of our integration tests run with the newer compiler version as well so we could replace it altogether (after more testing of course).

As the Arduino Uno Wifi Rev2 inbuilt LED is not on pin 13 but on 25 and 25 is not available on the board, I also added a new entry to the LED dropdown `BUILTIN_LED`, similar to the Arduino constant used in e.g. the Blink example.

boonto, 2019-12-03:
The avrdudes working with the Wifi Rev2 are also linked in the package index:
- [Linux ARM 32](http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino17-armhf-pc-linux-gnu.tar.bz2)
- [OSX](http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino17-x86_64-apple-darwin12.tar.bz2)
- [Linux 64](http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino17-x86_64-pc-linux-gnu.tar.bz2)
- [Linux 32](http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino17-i686-pc-linux-gnu.tar.bz2)
- [Windows](http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino17-i686-w64-mingw32.zip)

boonto, 2019-12-03:
Compilers for [OSX](http://downloads.arduino.cc/tools/avr-gcc-7.3.0-atmel3.6.1-arduino5-x86_64-apple-darwin14.tar.bz2), [Windows](http://downloads.arduino.cc/tools/avr-gcc-7.3.0-atmel3.6.1-arduino5-i686-w64-mingw32.zip) and [Linux ARM 32](http://downloads.arduino.cc/tools/avr-gcc-7.3.0-atmel3.6.1-arduino5-arm-linux-gnueabihf.tar.bz2) can also be found there.

boonto, 2019-12-03:
I tested the compiler and avrdude on Windows, both worked with standard Uno and Uno Wifi Rev 2. Additional compiler resources are not needed.

boonto, 2019-12-03:
Linux ARM compiler also seems to work, tested on a Raspberry Pi 4 and compiled for all avr-gcc robots.

boonto, 2019-12-04:
Connecting via Raspberry also works.

For the [IMU](https://www.arduino.cc/en/Guide/ArduinoUnoWiFiRev2#toc13) the additional library [Ardunio_LSM6DS3](https://www.arduino.cc/en/Reference/Arduino_LSM6DS3) is necessary.

boonto, 2019-12-05:
The previously mentioned library does not include the inbuilt temperature sensor, because of that the [library by SparkFun](https://github.com/sparkfun/SparkFun_LSM6DS3_Arduino_Library) will be used instead.

boonto, 2019-12-09:
The gyro and accelerometer blocks can be added to the program with CTRL-6. This will create the two program and config blocks.

bjost2s, 2020-01-02:
works fine with latest openroberta-connector

closed 2020-01-02 by bjost2s