OpenRoberta/openroberta-lab #333: Calliope: wrong error handling for ultrosonic sensor in simulation

bjost2s, 2019-11-26:
When you try to execute a program with the ultrasonic sensor in the simulation you get the following:
[image]
expected behaviour
[image]

As the configuration is still hidden (not present) it is not clear why we have a checker for this (at the wrong place)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
 - A clear and concise description of what you expected to happen.

**Screenshots**
 - If applicable, add screenshots to help explain your problem.

**Device information**
 - Type [Desktop/tablet/smartphone]
 - OS: [e.g. windows], including version
 - Browser [e.g. chrome, firefox], including version

**Additional context**
 - Add any other context about the problem here.

bjost2s, 2019-11-26:
Please check **all** sensors for calliope!

boonto, 2019-12-07:
Currently in progress by a GCI student.

boonto, 2019-12-08:
Has been resolved by a GCI student.

closed 2019-12-17 by boonto