OpenRoberta/openroberta-lab #334: SenseBox: Sound sensor returns wrong value

bjost2s, 2019-11-28:
**Describe the bug**
 Sound sensor returns simple analog read instead of the sound loudness.

**To Reproduce**
Create a program as in the screenshot and open the source code view

**Expected behavior**
Sensor returns the maximum absolute value from one wave period converted to % (0-100)

**Screenshots**
 - If applicable, add screenshots to help explain your problem.
[image]

**Additional context**
 see implementation for the sound sensor for Calliope

bjost2s, 2019-12-01:
The provided solution is not sufficient:
- In general we have decided not to add any more code to NEPODefs.h, desired place for new functions is inline, see #278 
- The calculation of max in the function reduces the range from the output, 0 (no noise) can never be reached.
-> We want to calculate the **peak amplitude** and convert it in percent: please improve the function, remove it from NEPODefs.h and create it like in #278 proposed

VinArt, 2019-12-02:
Moving to helper methods is currently impossible since helper methods for C based robots are not yet implemented

boonto, 2019-12-19:
Seems to work fine, constant noise, as well as short loud claps are detected.

closed 2019-12-19 by boonto