OpenRoberta/openroberta-lab #343: SenseBox: Nova SDS011 Feinstaub-Modul support

bjost2s, 2019-12-06:
more information here: https://shop.watterott.com/Nova-SDS011-Feinstaub-Sensor-Modul

VinArt, 2019-12-06:
- create a configuration block where the port is selectable: Serial1 or Serial2
- create a sensor block where the mode is selectable: PM2.5 or PM10
- provide code generation for configuration and for the sensor block

bjost2s, 2019-12-06:
ACCELEROMETER_TOOLTIP is declared twice in de.json and leads to an error

VinArt, 2019-12-10:
duplication removed in 879163d54fb1b58304975cd1543279135a1dc745

boonto, 2019-12-19:
Cannot test together with display as a line break is missing: [missingLineBreak.zip](https://github.com/OpenRoberta/openroberta-lab/files/3982778/missingLineBreak.zip)

boonto, 2019-12-19:
Seems to work, higher values are produced when dust is stirred up by beating a office chair.

closed 2019-12-19 by boonto