OpenRoberta/openroberta-lab #349: Gallery Loading

rishabhdeepsingh, 2019-12-09:
**Is your feature request related to a problem? Please describe.**
 - Whenever we open the gallery it takes some time to show up the results. While in the background it fetches for the files. But the user doesn't actually know that it is even loading.

**Describe the solution you'd like**
 - It will be great if we can implement some loading logo.

**Describe alternatives you've considered**
-  We can also reduce the requests to up to a single page only showing only around 50 results.

**Additional context**
 - Here is the result whenever we open up the gallery it takes around 11 minutes.
[image]

boonto, 2020-02-17:
rbudde improved the database calls, which should speed things up.
Additionally, the spinning logo was added with e1871d914c82d979bcd49b8379d8dde866dfe9cd.

rbudde, 2020-03-17:
works. The bottom text of the table, e.g. "Showing 1 to 11 of 11 rows" is bad. Instead of "rows" the text should use "items". Rows is ok in a traditional row format only.

bjost2s, 2020-03-22:
Not solved, spinning logo is still visible when gallery already is loaded.

boonto, 2020-04-02:
Spinning logo does not show up after the gallery is already loaded.

closed 2020-04-02 by boonto