OpenRoberta/openroberta-lab #351: EV3 and NXT simulation do not do the "motor on speed [number]% [rotation|degree]" block

philippmaurer, 2019-12-10:
**Describe the bug**
If you use the "motor on speed [number]% [rotation|degree]" block in the EV3 or NXT simulation, the wheel will only drive for a super short distance, no matter what degrees / rotations you enter.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a program that uses the "" block for NXT or EV3
2. Open the simulation and run your program
4. See error

**Screenshot**
[image]

**Expected behavior**
The robot should do what is says in the block.

corvus-albus, 2019-12-10:
Hi,
I'm not an developer but had a similar issue: https://github.com/OpenRoberta/openroberta-lab/issues/155 

To cite bjost2s:
> The simulation always uses the standard configuration (or less) of a robot system. You can not configure anything.
Once we decide (or someone provides) a full customizable simulation this will definitely a feature of this.
For now we offer the simulation only as a first step to try the lap. Once you really want to go deeper into programming you should use the real robot.

I think you have to deal with it. In fact I'm so happy to have a simulation for teaching that it is easy for me to accept it.
Cheers corvus-albus

boonto, 2019-12-10:
Yes the B <-> C thing with the motors is intended behavior for now.

The degree thing seems to be a bug though.

closed 2019-12-20 by VinArt