OpenRoberta/openroberta-lab #352: NXT simualtion get stuck with TypeError: "navigator.mediaDevices.getUserMedia is not a function"

corvus-albus, 2019-12-10:
**In order for us to understand your problem better, please provide us with some information**
Please provide relevant information, for example if it is a connection issue, tell us which robot you are using, which version of USB program you are running on which OS.

Another example: if it is a custom installation problem, please tell us which hardware you are running and what exactly failed for you.

boonto, 2019-12-10:
This happens due to the "unsecured" connection to the "remote" server. The `getUserMedia()` method will not work ([see documentation](https://developer.mozilla.org/en-US/docs/Web/API/MediaDevices/getUserMedia)). The error also occurs for Calliope, but we catch it and display a message:
```
Sorry, but there is no microphone available on your system
```
We should catch this error for NXT as well and display the same message.

corvus-albus, 2019-12-10:
> We should catch this error for NXT as well and display the same message.

That would be great.

Anyway: Is there an easy way to establish a secure connection from within openroberta-server or do I have to fiddle with apache or nginx?

boonto, 2019-12-10:
There are almost identical implementations (except for the error catching) in `robot.nxt.js` and `robot.calliope.js`, we should probably pull it up to reduce duplication.

boonto, 2019-12-10:
You would have to set up HTTPS on your raspberry, but there seems to be a workaround for Chrome: [see answers of this Stack Overflow question](https://stackoverflow.com/questions/34165614/navigator-mediadevices-getusermedia-is-not-working-and-neither-does-webkitgetuse).

corvus-albus, 2019-12-10:
>There are almost identical implementations (except for the error catching) in robot.nxt.js and robot.calliope.js, we should probably pull it up to reduce duplication.

Many thanks for this hint. As a fast (local) fix I copied the corresponding lines from robot.calliope.js to robot.nxt.js and now it works fine! 

>You would have to set up HTTPS on your raspberry, but there seems to be a workaround for Chrome: see answers of this Stack Overflow question.

Thank you you for this as well. I tried but failed. I have to dig a little bit deeper. I will ask my colleagues for help ...

Cheers 
corvus-albus

closed 2019-12-20 by VinArt