OpenRoberta/openroberta-lab #354: Simulation stops running if variable reaches 0

boonto, 2019-12-10:
**Describe the bug**
 - The simulation stops running if a variable reaches 0 in combination with the drive block (or any block?)
- This does not happen if the variable never reaches 0 (change decrement in program to e.g. 3, the robot will reverse)

**To Reproduce**
Steps to reproduce the behavior:
1. Import [this program](https://github.com/OpenRoberta/openroberta-lab/files/3946919/simulation_stops.zip)
2. Run in simulation
3. Observe robot and program stopping

**Expected behavior**
 - The simulation keeps on running and the robot will reverse.

**Screenshots**
[image]

rbudde, 2019-12-11:
division by 0 (speed) lead to NAN. This NAN was used as prediction for the next "wakeup" of the simulator and that, in turn, lead to a freeze. More errors of this kind removed, too.

boonto, 2019-12-17:
Simulation does not freeze anymore.

closed 2019-12-17 by boonto