OpenRoberta/openroberta-lab #359: Break out of loop block outside of loop produces server error

boonto, 2019-12-12:
**Describe the bug**
- Using the "break out of loop" block outside of a loop produces a server error.
- A warning is shown on the block but it does not prevent the server error

**To Reproduce**
Steps to reproduce the behavior:
1. Use break out of loop block outside of a loop
2. Open simulation, source code view or similar actions
4. See error

**Expected behavior**
Either
- You get notified there are errors in your program with a popup, or
- a warning is shown on the block and and it is ignored in further processing

**Screenshots**
[image]

boonto, 2019-12-12:
In default blockly it is disabled if it is outside of a loop, for our blockly this does not seem to be the case. If it is disabled it will be ignored by the code generation and no error will be generated.

I think it was disabled in our blockly as well at some point and this probably changed.

boonto, 2019-12-13:
I tested the check similar to [original blockly]. It works, but there are still some visual errors with the block being active but not shown as active.

boonto, 2019-12-16:
Works with the following changes in loops.js:
```
...
     } while (block);
     if (legal) {
       this.setWarningText(null);
+      this.setDisabled(false);
     } else {
       this.setWarningText(Blockly.Msg.CONTROLS_FLOW_STATEMENTS_WARNING);
+      if (!this.isInFlyout && !this.getInheritedDisabled()) {
+        this.setDisable(true);
+      }
     }
...
```
@bjost2s what do you think? Is there any reason not to do this?

VinArt, 2019-12-20:
Block becomes inactive outside the loop and active back when put into loop.

closed 2019-12-20 by VinArt