OpenRoberta/openroberta-lab #360: Some Calliope blocks produce server errors in simulation

boonto, 2019-12-13:
**Describe the bug**
- The blocks "set LED Bar", "clear 4 Digit Display" and "set 4 Digit Display" (maybe more, will test) produce a server error when opening the simulation. 

**To Reproduce**
Steps to reproduce the behavior:
1. Add one of the mentioned blocks
2. Open simulation
4. See error

**Expected behavior**
 - No errors, a warning on the block that it cannot be used in simulation.

**Additional context**
 - Should be covered by workflows integration tests.

boonto, 2019-12-13:
In this context I will create Calliope integration tests with all action and sensor blocks.

boonto, 2019-12-13:
There already are fractured integration tests for the blocks, but the robot specific ones do not have the workflow tests. I'll add the workflow tests to the robot specific tests and combine the fractured tests into one with all blocks.

boonto, 2019-12-16:
Workflow and simulation tests will be tracked in #364.

VinArt, 2019-12-20:
Now warning signs are shown, but no server error.

closed 2019-12-20 by VinArt