OpenRoberta/openroberta-lab #362: Add "create list with x repeated" block to Calliope

boonto, 2019-12-13:
**Is your feature request related to a problem? Please describe.**
 - Most other robots that have list capabilities have the block "create list with x repeated" block, this is missing in Calliope

**Describe the solution you'd like**
 - The block added to the Calliope toolbox

nishanth1232, 2019-12-24:
https://github.com/OpenRoberta/openroberta-lab/pull/410

boonto, 2019-12-30:
Pull requests #436 and https://github.com/OpenRoberta/ora-cc-rsc/pull/2 by a GCI student resolved this. Will probably be merged after the release.

boonto, 2020-01-15:
Merged in https://github.com/OpenRoberta/openroberta-lab/commit/66d90d9235b162619b3283b9ccba3674080b7e1f and https://github.com/OpenRoberta/ora-cc-rsc/commit/cad89d8d7576554e28f933006028240c33fc6250.

closed 2020-01-15 by boonto