OpenRoberta/openroberta-lab #364: Add simulation tests for all common & robot specific programs

boonto, 2019-12-13:
Currently there are multiple types of compiler integration tests:
- common integration tests which go over the REST compile call with all suitable programs
- robot specific integration tests which go over the REST compile call with all suitable programs
- workflow tests which test every workflow of a robot with an empty program

Instead the tests can be restructured to
- [ ] Test each REST (compiler workflow) call once?
- [ ] Test with the `compile` and `getsimulationcode` workflows instead of the previous compile REST call
- [ ] Test remaining workflows with empty programs

This would simplify the tests and add some test coverage for the simulation and prevent issues like #322 and #357.

boonto, 2020-01-27:
I've found some smaller issues during this which I fixed immediately:
- "get random element from list" blocks did generate invalid code in C++ and Java. There is still no implementation now as we wanted to remove that block anyway. It just returns the first element now, but compiles at least
- Java "create with list repeated" could not accept double variables. Value is cast to int now
- Java "get sub list" could not accept double variables. Values are cast to int now
- Connection type lead to server error when using simulation
- RFID sensor in wait until block lead to server error
- Uno Wifi Rev.2 digital & analog write could not work with variables. Value is cast to int now
- Bot'n Roll could not output light sensor values
- Edison if return could not be used
- NXT show text could not be used with boolean variables

For all bigger issues I created specific issues and disabled the blocks in the "allBlocks" programs.
Programs that have disabled blocks in them but succeed otherwise are marked as "part" for partial success. These partially successful tests need to be looked at in the respective issues I created.

The tests run in parallel across robots for now and still add 6 minutes to the integration tests. I didn't run into any problems here, but we should keep an eye out in the future. This should speed up significantly with the precompiled Arduino binaries.

rbudde, 2020-01-28:
only #711 remains to be done

closed 2020-01-28 by rbudde