OpenRoberta/openroberta-lab #365: Machine Learning

rao003, 2019-12-14:
Have you seen this feature ? With API targeting some IBM servers, they have made extra blocks for Scratch, with Machine Learning Models ... that would be VERY COOL in Open Roberta !!!

https://machinelearningforkids.co.uk/#!/worksheets

[image]
[image]

bjost2s, 2019-12-14:
Thank you very much! We are aware of this. Still trying to find a way to bring the models to the (small) devices that we support. AFAIK all models are running on the PC's or servers, not on the robots. We are happy about help!

rao003, 2019-12-15:
Hallo Beate,

what about using the API from IBM or [Google Colab](https://colab.research.google.com/notebooks/welcome.ipynb)
at least on the web version ? Let the big players do the heavy work for you....

Mfg
RAO

quote
 Built by [Dale Lane](http://dalelane.co.uk/) using APIs from [IBM Watson](https://www.ibm.com/watson/developer/).

For more details about the implementation, you can see the source code on [GitHub](https://github.com/IBM/taxinomitis/), or read some of the blog posts about the tech.
unquote