OpenRoberta/openroberta-lab #377: EV3: Compiler error due to order of function definitions

bjost2s, 2019-12-17:
Attention, this error only appears for C4EV3, EV3Dev and Lejos works fine!
this works:
[image]
this doesn't
[image]

**Expected behavior**
 Compilation succeeds.

Function declaration may help here.

RohanCheri, 2019-12-29:
Oh I used the wrong Issue number my bad

RohanCheri, 2019-12-29:
I meant to put Issue #390

Blippy2019, 2020-01-02:
Hi, I was looking at this issue and tested it on the lab, test server and my local one but the issue didn't come up. I used the exact same code and tried it both chrome and firefox with the C4EV3. Wondering if this issues is already fixed?
[image]

boonto, 2020-01-02:
It cannot be compiled if you try to compile the program for the real robot. As can be seen in the source code view "macheEtwas2" is used inside "macheEtwas" before it is declared.

boonto, 2020-01-06:
Resolved by GCI student.

closed 2020-01-06 by boonto