OpenRoberta/openroberta-lab #437: NXT: Robot still moves forward with negative distance value

Blippy2019, 2019-12-29:
**Describe the bug**
 - When there is a negative distance input for the "steer ... speed left ... speed right ... distance ..." block, the simulation bot moves forward slightly. 

**To Reproduce**
Steps to reproduce the behavior:
1. Find the steer block that takes in distance.
2. Set left motor and right motor to 100% and distance to any negative number
3. Run the simulation
4. See error

**Expected behavior**
 - Robot moves backwards or alert message of "no negative numbers"

**Screenshots**
[image]

**Device information**
 - Type Desktop
 - OS: Windows 10
 - Browser: Chrome 

**Possible fix**
 - There is no specifications on the range for this parameter, nothing says the value cannot be negative, maybe add information on the valid range in help information.

boonto, 2020-01-03:
Needs testing whether this is the case for the real robot as well.

boonto, 2020-01-03:
The real robot does go forward if both of the speed values are different. It goes backwards when they are the same.

boonto, 2020-01-15:
Partially resolved by a GCI student. The robot behaves as expected in the simulation.