OpenRoberta/openroberta-lab #439: EV3: Setting volume to 0 does not decrease the volume

Blippy2019, 2019-12-29:
**Describe the bug**
 - Setting the volume to 0 % does not do anything to the EV3 sound, it sounds the same as when the volume is 100 %

**To Reproduce**
Steps to reproduce the behavior:
1. Find the "set volume %" block
2. Set input to 0 %
3. Play any sound
4. See error

**Expected behavior**
 - There should be no sound output as the volume is set to 0%

**Screenshots**
 - 
[image]
[image]

**Device information**
 - Type: Desktop
 - OS: Windows 10
 - Browser: Chrome

**Additional context**
 - According to the help information, 0% would be no sound at all.

boonto, 2020-01-03:
Needs testing whether this is the case for the real robot as well.

boonto, 2020-02-17:
Volume does not work because `GainNode` is not used, instead the oscillator is connected directly (See [this](https://stackoverflow.com/questions/43386277/how-to-control-the-sound-volume-of-audio-buffer-audiocontext) and [this](https://developer.mozilla.org/en-US/docs/Web/API/Web_Audio_API)).
Additionally, a volume of `0` is not evaluated correctly in an if, fixed when merging #518.

closed 2021-08-12 by bjost2s