OpenRoberta/openroberta-lab #450: Arduino: wrong code generation for infrared, ultrasonic and rfid sensors

bjost2s, 2019-12-30:
**Describe the bug**
 see screenshot, it is obvious that always one of the two sensor values are returned.

[image]

**Expected behavior**
The corresponding sensor (identified via the name given in the configuration) should be passed to functions.

**Additional context**
Related to #306!

bjost2s, 2019-12-30:
related to #278, don't forget to integrate this corrected functions later into the yaml files ;-)

bjost2s, 2020-01-02:
The fix works for ultrasonic and rfid. The infrared sensor still doesn't work properly with more than one sensor attached, might be a problem of the library, see https://arduino.stackexchange.com/questions/52090/can-i-connect-more-than-one-ir-receiver-to-an-arduino-uno e.g.

boonto, 2020-01-07:
Generated code contains the different sensors now.

closed 2020-01-07 by boonto