OpenRoberta/openroberta-lab #455: EV3: Say block doesn't work with some inputs

Blippy2019, 2019-12-31:
**Describe the bug**
 - When using the say block, some values won't work, such as 0, empty text, false. When these inputs are added, the program will still run but never finish. Any sensor with a false value also won't work.

**To Reproduce**
Steps to reproduce the behavior:
1. Add a "say" block
2. Set the parameter to any of the mentioned values
3. Run the program
4. See error

**Expected behavior**
 - Program says the given input

**Screenshots**
 - 
[image]
None of the say blocks in the above screenshot works, the program never finishes and the bot never plays the C note.

**Device information**
 - Desktop
 - Windows 10
 - Chrome

**Additional context**
 - The program can say True but not False
 - Connecting the block "wait for connection" to the "say" block generates a server error 
 - There is no information on what inputs can be attached to the "say" block

boonto, 2020-01-03:
Needs testing whether this is the case for the real robot as well.

boonto, 2020-01-14:
Resolved by a GCI student.

closed 2020-01-14 by boonto